waveshareteam/ESP32-display-support #7: Using PSRAM and Buzzer

jkunkee, 2025-03-01:
Hi!

I'm a hobbyist hoping to use my ESP32-S3 1.69inch Display Development Board to drive my air quality analyzer. As part of this, I was considering putting together a complex GUI. I hit a snag, though: if I enable the PSRAM, the Buzzer clicks randomly.

Looking at the datasheet, it seems the ESP32-S3-R8 has in-package octal SPI PSRAM, so GPIO33 is SPIIO4 in that configuration. The data sheet also calls the R8 in-package PSRAM an octal SPI peripheral. So I thought I'd ask here: do you know if there a way to configure the ESP32 to successfully use the in-package octal SPI PSRAM as a quad SPI peripheral? That should avoid the GPIO33 conflict.

Thanks,
Jon

WaveshareCloud, 2025-03-01:
Hi @jkunkee ,
Can you take a picture of the back of the board? I need to differentiate between versions

jkunkee, 2025-03-01:
I thought it was a V1 because it lacks the V2 sticker, but now that you ask I haven't checked any further. Picture:

[image]

WaveshareCloud, 2025-03-01:
OK，It is difficult to judge from the appearance, so when did you buy this board? If you have tested the functionality of the buzzer and key using the example below, you can tell the difference between the GPIO allocation.
If you have an older version, you may need to remove a few keys to disable the buzzer, and there is no way to customize the PWRKEY button
https://www.waveshare.com/wiki/ESP32-S3-Touch-LCD-1.69#09_LVGL_Keys_Bee

WaveshareCloud, 2025-03-01:
Ok, a long time ago. First, let's confirm the final requirements. For the V1 version of hardware, you can remove one resistor and one transistor as shown in the figure below, so that you will not be able to use the buzzer, and the buzzer will not consume power.
Please make sure this is what you need before removing the device. For the hardware design issues, I'm sorry for giving you a bad experience. This has been fixed for V2, so if you want to buy another piece, don't worry.
[image]

jkunkee, 2025-03-01:
It's fascinating how "a long time ago" can mean so many different things, especially in the technology world.

Thank you for the quick replies and straightforward instructions! I'll look at my software needs a bit closer and, if truly necessary, I can remove the buzzer or buy a new piece.

I consider my issue resolved.

Have you ever had anyone use 80MHz SPI and GPIO33 to try to do pulse density modulation to play full spectrum audio through the buzzer? I can't imagine it would work well, but it would be fascinating to get it to work. It would probably be better with SPI2 since that can mux to any GPIO and has a dedicated DMA channel...

WaveshareCloud, 2025-03-03:
Hi @jkunkee ,
I haven't met anyone sharing them yet.

closed 2025-03-16 by jkunkee

jkunkee, 2025-07-09:
@WaveshareCloud , I'm just getting to this modification and it looks like my V1 board is different from the board you annotated. The set of 2-pin components only has three components, and there is a fourth transistor. Would you be willing to confirm for me the location of R14 (the GPIO-to-base-current-limiting transistor for the buzzer) on my version of the board?