waveshareteam/ESP32-display-support #9: LCD Backlight Troubleshooting

jkunkee, 2025-03-16:
Hi! I'm working on standing up Tasmota on the ESP32-S3 1.69-inch Display Development Board (V1) for my own use. (My DIY notes are shared in PR #6.) Tasmota is a large PlatformIO project that leverages the ESP-IDF, and I've been using example 01_ESP_IDF_ST7789 from the wiki to double-check my configuration.

Using Tasmota's interface to toggle GPIO15 only slightly changes the backlight's brightness instead of turning Q1 all the way off. Does that sound like an issue you've seen before, like a GPIO misconfiguration? If it doesn't sound familiar, is there a way I could find out where on the board I could probe GPIO15, the base of Q1, or R16?

Y1hsiaochunnn, 2025-03-24:
Normally, it can be controlled by controlling the GPIO15 output PWM. Of course R16 is here, you can also replace it with a larger resistor

[image]

jkunkee, 2025-03-27:
That helps a lot. Thanks!

closed 2025-03-27 by jkunkee

jkunkee, 2025-04-03:
For anyone else coming along with the same question, it seems that R11's value and R16's value combined with the `0.1*VDD` GPIO output voltage mean that GPIO15 going low cannot drive Q1 into the cutoff region. Instead it stays in the linear region, apparently not far from saturation. I don't have rework equipment fine enough to replace R11, but I'm considering simply removing it since Q1 is a BJT and not a FET, the ESP32-S3 GPIOs seem fairly sane, and R16 will current-limit the Q1 input to about 3mA (not too far, I hope, from the design I_BE of ~2.2mA).