waveshareteam/ESP32-display-support #12: ESP32-P4-Module-DEV-KIT with 7-DSI-TOUCH-A

Speedy2k, 2025-05-08:
It would be very nice to have a sample code to use those devices togheter since you sold them as a kit .
I would like to have sample code for arduino and or Platform IO i currently us PIOArduino since Platform IO doesn't support the latest ESP32 arduino core.

We need good device initializations, hardware rotation, backligth control. 

Thanks a lot!

Y1hsiaochunnn, 2025-05-09:
Hi @Speedy2k ,

It is my impression that the 7-DSI-Touch-A does not support hardware rotation.

Testing the program on Arduino is OK, but the code is not released yet due to some i2c compatibility issues.

PlatformIO also works, but it's not our main target for now.

I'll let you know when the Arduino program updates, but it will take some time.

Speedy2k, 2025-05-09:
Do you have any test samples that i can help you with ? I have been able to control de backlight touch on arduino and even get LVGL to work. But i got problem when rebooting the esp32, when i completely unpluged the esp32 and replug it the screen show up the backligth work perfectly and the touch also. but when i do a software reset on the esp32, the touch work, the UI is loading, the backligth is working but nothing show on the display

Y1hsiaochunnn, 2025-05-09:
This sounds like a MIPI signal disorder caused by the screen not being reset. You can reset the screen.

Something like what I've defined here
https://github.com/waveshareteam/Waveshare-ESP32-components/blob/958e50b80bd34b43ea3bfafc8e01aeacd08dd3c3/display/lcd/esp_lcd_ili9881c/esp_lcd_ili9881c.c#L134-L150

[image]

Speedy2k, 2025-05-09:
Nice it worked fine now the LCD is restarting after a soft reboot and and hard reboot. I have made a small initialisation librairy for this screen. Now it work well. I am adding my files if it can help for an even better solution. Right now it use the Esspresiff ESP32_Display_Panel librairie. The mirrors are working but can't seems to rotate the screen. IT must be doable because the software rotate slow the rendering by a lot!!!

[ILI9881C_LCD.ZIP](https://github.com/user-attachments/files/20129190/ILI9881C_LCD.ZIP)

WaveshareCloud, 2025-05-21:
Once the ESP32_Display_Panel has adapted to the latest i2c_master driver, I will adapt it to the pr screen

closed 2025-05-21 by WaveshareCloud

Y1hsiaochunnn, 2025-06-30:
@Speedy2k Now that we have completed a relatively rough example on Arduino, you can download the demo from our Wiki if you are interested. The demos are powered by GFX library

Speedy2k, 2025-07-01:
Perfect! I will try that!

Y1hsiaochunnn, 2025-07-02:
What version of Arduino do you use? Is PSRAM enabled?

Speedy2k, 2025-07-02:
Arduino 1.8.19 with arduino Core 3.2.0 and yes PSRAM is activated, Flash Mode i have tried both and neither are working

[image]

Y1hsiaochunnn, 2025-07-03:
Sorry, I can't confirm what the problem is and there is no log output? Currently I am working well on Arduino-esp32 v3.2.0 version