waveshareteam/ESP32-display-support #13: How to change the IP on the ESP32 Board

josyulakp, 2025-05-09:
Hello, 

1) I am using this waveshare board shown in the picture (https://www.waveshare.com/servo-driver-with-esp32.htm
) I'm wondering how to change the IP of this board.
2) Where I do find the firmware to reflash the board I have uploaded a program from Arduino IDE and it turned of the LCD board and not broadcasting the hotspot anymore. 

Requesting immediate assistance, thank you very much. 

[image]

peekaboo29, 2025-05-10:
1. This board cannot change the IP address. 2. The OLED works by burning this program, and the version of the ESP32 development kit is 1.0.6. Burning other servo control example programs only simply drives the servo and does not include the program for the OLED display.

[image]

josyulakp, 2025-05-10:
@peekaboo29  thank you very much for the response, I need to access 2 boards at the same time and _set all the motors to servo_ mode and _enable serial forwarding_, currently the issue is since both boards have the same IP I am not able to connect to them individually to enable these settings. Can you please guide me how to do this?

WaveshareCloud, 2025-05-21:
I think the problem you encountered is that two development boards enable AP mode at the same time, and then share two WiFi. Under the existing logic, you need to connect them separately and control them?

If so, I'd rather use STA mode and then connect to the same network, like this:

Device 1 access WiFi router IP assumption: 192.168.xx.101
Device 2 access WiFi router IP assumption: 192.168 xx. 102
Mobile phone access WiFi router IP assumptions:192.168 xx. 103

Then, you can use your phone to access different device IP addresses for individual control.

A more convenient form is to use mqtt to control different devices.

So the above problem seems to be more of a network protocol problem, and a control side problem. You can try my suggestions first. This has nothing to do with the driver part and I will close this issue

closed 2025-05-21 by WaveshareCloud