waveshareteam/ESP32-display-support #14: esp32 1.69 oled

bjwcleary, 2025-05-14:
I have the watch programed and working perfectly while on usb power but when I connect the battery the screen won't stay on unless I press and hold the pwr button what am I missing??

Y1hsiaochunnn, 2025-05-15:
maybe you should check this features
https://www.waveshare.com/wiki/ESP32-S3-Touch-LCD-1.69#09_LVGL_Keys_Bee
[image]

bjwcleary, 2025-05-15:
yeah I've seen that but unsure how to use it.

bjwcleary, 2025-05-15:
what do you mean have you any examples?

Y1hsiaochunnn, 2025-05-15:
[image]

WaveshareCloud, 2025-05-21:
please use arduino demo 09_LVGL_Key_Bee, you'll get the conclusion

closed 2025-05-21 by WaveshareCloud