waveshareteam/ESP32-display-support #15: ESP32-S3-Touch-LCD-4 support

dinisfc, 2025-05-27:
Currently having a really hard time trying to make use of this display. Using the official ESP libraries is a pain as each major update seems to break everything, including config files that need to be changed manually. Is it possible to have full official support for this board? I've been trying to get it running LVGL with a I2C sensor for months now and everything is constantly breaking. The hardware already seems unstable and unreliable by itself, even harder when I'm constantly troubleshooting library incompatibilities and bugs.

Y1hsiaochunnn, 2025-05-27:
Hi @dinisfc ,
I'm sorry this isn't working so well, but I recommend that you don't use ESP32_Panel_Display until it stabilizes to a version, or keeps the version unchanged.

First of all, Waveshare Wiki provides some examples of GFX driver library routines, touch part, you can use Sensorlibs, fixed arduino-esp32 version, currently arduino-esp32 iteration, new version of the driver is constantly updated, if you want to sync the version, There are certainly problems with the driver. Even the library maintenance and developers of Espressif cannot keep up with the update speed of arduino-esp32 in time. You can see this in the issues that were communicated today

Secondly, using the BME680 driver is through the i2c driver, you can share the driver library you use, let me be more specific to determine the problem.

If there is a problem with the hardware, please share it, and I will help you eliminate it

Y1hsiaochunnn, 2025-06-23:
inactivity and closed

closed 2025-06-23 by WaveshareCloud