waveshareteam/ESP32-display-support #16: ESP touch LCD 4inch Rev 3.0 backlight causes brownout

pmarches, 2025-07-01:
When the backlight of the 4inch display is turned off for a few seconds, the display shuts down with a brown out error.

Here is a sample program that demonstrates the issue. The IO multiplexer is initialized, all output pins set to LOW. It then toggles the beeper to show it is running. After a few seconds the display turns off.
 
`
#include <cstddef>
#include <esp_log.h>
#include <esp_err.h>
#include <driver/i2c.h>

#define TAG __FILE__

extern "C" void app_main(void) {
    i2c_config_t i2c_conf = {
        .mode = I2C_MODE_MASTER,
        .sda_io_num = GPIO_NUM_15,
        .scl_io_num = GPIO_NUM_7,
        .sda_pullup_en = GPIO_PULLUP_ENABLE,
        .scl_pullup_en = GPIO_PULLUP_ENABLE,
        .master={.clk_speed = 400000},
    };

    i2c_param_config(I2C_NUM_0, &i2c_conf);
    ESP_ERROR_CHECK(i2c_driver_install(I2C_NUM_0, i2c_conf.mode, 0, 0, 0));

#define TCA9554_I2C_ADDRESS 0x20
#define I2C_MASTER_TIMEOUT_MS       1000

#define TCA9554_REG_INPUT 0x00
#define TCA9554_REG_OUTPUT 0x01
#define TCA9554_REG_INVERSION 0x02
#define TCA9554_REG_CONFIG 0x03

#define TOUCHPAD_RESET_PIN (1u << 0)
#define BACKLIT_PIN (1u << 1)
#define LCD_RESET_PIN (1u << 2)
#define BEEPER_PIN (1u << 5)
#define RTC_INT (1u << 6)
#define CHARGER_IRQ (1u << 7)
#define OUTPUT_PINS (TOUCHPAD_RESET_PIN|BACKLIT_PIN|LCD_RESET_PIN|BEEPER_PIN)

    uint8_t write_buf[]={TCA9554_REG_OUTPUT, 0|BACKLIT_PIN}; //Important to leave the backlit pin ON
    ESP_ERROR_CHECK(i2c_master_write_to_device(I2C_NUM_0, TCA9554_I2C_ADDRESS, write_buf, 2, pdMS_TO_TICKS(I2C_MASTER_TIMEOUT_MS)));

    write_buf[0] = TCA9554_REG_CONFIG;
    write_buf[1] = (uint8_t) ~OUTPUT_PINS; //configure some pins as output, input is bit 1
    ESP_ERROR_CHECK(i2c_master_write_to_device(I2C_NUM_0, TCA9554_I2C_ADDRESS, write_buf, 2, pdMS_TO_TICKS(I2C_MASTER_TIMEOUT_MS)));
#if 0
#define SW6106_I2C_ADDRESS 0x3C
#define SW6106_LIGHT_LOAD_REG 0x38
#define SW6106_LIGHT_LOAD_DISABLE 0x0A
    write_buf[0] = SW6106_LIGHT_LOAD_REG;
    write_buf[1] = SW6106_LIGHT_LOAD_DISABLE;
    ESP_ERROR_CHECK(i2c_master_write_to_device(I2C_NUM_0, SW6106_I2C_ADDRESS, write_buf, 2, pdMS_TO_TICKS(I2C_MASTER_TIMEOUT_MS)));
#endif

    bool high_level=true;
    uint8_t selected_pin = BEEPER_PIN;
    while(true){
        uint8_t original_reg_value;
        write_buf[0]=TCA9554_REG_INPUT;
        ESP_ERROR_CHECK(i2c_master_write_read_device(I2C_NUM_0, TCA9554_I2C_ADDRESS, write_buf, 1, &original_reg_value, 1, pdMS_TO_TICKS(I2C_MASTER_TIMEOUT_MS)));
        ESP_LOGW(TAG, "original_reg_value=0x%x", original_reg_value);

        high_level=!high_level;
        if(high_level){
            write_buf[1]=original_reg_value|selected_pin;
        }
        else{
            write_buf[1]=original_reg_value & ~selected_pin;
        }

        write_buf[0]=TCA9554_REG_OUTPUT;
        ESP_ERROR_CHECK(i2c_master_write_to_device(I2C_NUM_0, TCA9554_I2C_ADDRESS, write_buf, 2, I2C_MASTER_TIMEOUT_MS / portTICK_PERIOD_MS));

        ESP_LOGI(TAG, "Wait..");
        if(high_level){
            vTaskDelay(pdMS_TO_TICKS(10));
        }
        else{
            vTaskDelay(pdMS_TO_TICKS(1000));
        }
    }
}

`

Y1hsiaochunnn, 2025-07-03:
Here, waveshare mentions this issue in the FAQ of the Wiki. But as far as I know, they are doing V4.0 on this board because of the discontinuation of SW6106.

[image]

Y1hsiaochunnn, 2025-07-04:
In fact, it is caused by the light load of SW6106. I do have a register list here, but it is in Chinese, and I have not found the English version

[SW6106_寄存器列表_Release_RG005_1_v1.0.pdf](https://github.com/user-attachments/files/21049051/SW6106_._Release_RG005_1_v1.0.pdf)

pmarches, 2025-07-05:
The FAQ suggestion does match with the datasheet register list. I tried again, but with the same brownout result. The datasheet makes this mention: "A-port No-Load Detection Current Threshold when output resistance = 30mΩ"  I do not have a battery attached to the display. Could this be a problem?

```
#include <esp_log.h>
#include <esp_err.h>
#include <driver/i2c.h>

#define TAG __FILE__

void app_main(void) {
    i2c_config_t i2c_conf = {
        .mode = I2C_MODE_MASTER,
        .sda_io_num = GPIO_NUM_15,
        .scl_io_num = GPIO_NUM_7,
        .sda_pullup_en = GPIO_PULLUP_ENABLE,
        .scl_pullup_en = GPIO_PULLUP_ENABLE,
        .master={.clk_speed = 400000},
    };

    i2c_param_config(I2C_NUM_0, &i2c_conf);
    ESP_ERROR_CHECK(i2c_driver_install(I2C_NUM_0, i2c_conf.mode, 0, 0, 0));

#define TCA9554_I2C_ADDRESS 0x20
#define I2C_MASTER_TIMEOUT_MS       1000

#define TCA9554_REG_INPUT 0x00
#define TCA9554_REG_OUTPUT 0x01
#define TCA9554_REG_INVERSION 0x02
#define TCA9554_REG_CONFIG 0x03

#define TOUCHPAD_RESET_PIN (1u << 0)
#define BACKLIT_PIN (1u << 1)
#define LCD_RESET_PIN (1u << 2)
#define BEEPER_PIN (1u << 5)
#define RTC_INT (1u << 6)
#define CHARGER_IRQ (1u << 7)
#define OUTPUT_PINS (TOUCHPAD_RESET_PIN|BACKLIT_PIN|LCD_RESET_PIN|BEEPER_PIN)

#define SW6106_I2C_ADDRESS 0x3C
#define LIGHT_LOAD_REG 0x38
    uint8_t write_buf[2];

    write_buf[0]=LIGHT_LOAD_REG;
    uint8_t read_buf[1]={0xfe};
    ESP_ERROR_CHECK(i2c_master_write_read_device(I2C_NUM_0, SW6106_I2C_ADDRESS, write_buf, 1, read_buf, 1, pdMS_TO_TICKS(I2C_MASTER_TIMEOUT_MS)));
    ESP_LOGE(TAG, "SW6106 LIGHT_LOAD_REG was 0x%x", read_buf[0]);

    write_buf[1]=0x0A;
    ESP_ERROR_CHECK(i2c_master_write_to_device(I2C_NUM_0, SW6106_I2C_ADDRESS, write_buf, 2, pdMS_TO_TICKS(I2C_MASTER_TIMEOUT_MS)));

    write_buf[0]=TCA9554_REG_OUTPUT;
    write_buf[1]=0;
    bool keep_backlight_on=false;
    if(keep_backlight_on){ //Try this code with true and false
        write_buf[1]=BACKLIT_PIN;
    }
    ESP_ERROR_CHECK(i2c_master_write_to_device(I2C_NUM_0, TCA9554_I2C_ADDRESS, write_buf, 2, pdMS_TO_TICKS(I2C_MASTER_TIMEOUT_MS)));

    write_buf[0] = TCA9554_REG_CONFIG;
    write_buf[1] = (uint8_t) ~OUTPUT_PINS; //configure some pins as output, input is bit 1
    ESP_ERROR_CHECK(i2c_master_write_to_device(I2C_NUM_0, TCA9554_I2C_ADDRESS, write_buf, 2, pdMS_TO_TICKS(I2C_MASTER_TIMEOUT_MS)));

    while(true){
        ESP_LOGI(TAG, "Wait..");
        vTaskDelay(pdMS_TO_TICKS(1000));
    }
}

```

ramzan-zz, 2025-07-21:
unrelated to your problem, but is it possible if anyone has any insight on how to have the board powered on automatically when connected via 7-36v terminals without clicking the pwr key on rev3.0?

pmarches, 2025-07-30:
> The FAQ suggestion does match with the datasheet register list. I tried again, but with the same brownout result. The datasheet makes this mention: "A-port No-Load Detection Current Threshold when output resistance = 30mΩ" I do not have a battery attached to the display. Could this be a problem?

Is a battery mandatory if I want to turn the screen off?

Y1hsiaochunnn, 2025-07-31:
It is recommended to use battery.

WaveshareCloud, 2025-08-28:
> unrelated to your problem, but is it possible if anyone has any insight on how to have the board powered on automatically when connected via 7-36v terminals without clicking the pwr key on rev3.0?

As mentioned above, rev4.0 supports this. The current version of the design must be initiated by pressing the key.

closed 2025-08-28 by WaveshareCloud